piface/PiFace-Real-Time-Clock #1: Not working with Raspberry Pi 2

benoitguigal, 2015-05-27:
I can't get this script to work on a Raspberry Pi 2. 
Any idea ? 
Thanks a lot

tompreston, 2015-06-11:
What happens when you run it? Does it error?

benoitguigal, 2015-07-23:
Hello sorry for late reply. 
When the Raspberry Pi boots I get the following error 
```
/etc/rc.local: 23 /etc/rc.local: cannot create /sys/class/i2c-dev/i2c-1/device/new_device: Directory nonexistent
```

tompreston, 2015-07-24:
What do you get from the output of:

    ls -l /sys/class/i2c-dev/

The install script adds the following lines to `/etc/rc.local`. The problem line is the third one below:

    modprobe i2c-dev
    modprobe i2c:mcp7941x
    echo mcp7941x 0x6f > /sys/class/i2c-dev/i2c-$i/device/new_device
    hwclock -s

Where `$i` is either 1 or 0 depending on which Raspberry Pi you have.

benoitguigal, 2015-07-24:
Actually I forgot to enable i2c in raspi-config prior to running the script. Maybe the error was due to that ?

tompreston, 2015-07-24:
Since you hadn't enabled i2c in raspi-config then the device wouldn't appear which would explain the error. Are you still having problems? Does the RTC work even though you're getting the error?

benoitguigal, 2015-07-24:
I managed to have it worked after enabling i2c, thanks

closed 2015-07-24 by benoitguigal

nemequ, 2015-07-28:
It would be nice if the script would output better error message.  A simple

```
if [ ! -e /sys/class/i2c-dev/i2c-$i ]; then
  echo "Error: you must enable i2c (use \`raspi-config\`, the \"Advanced Options\" section)."
fi
```

should (untested) do the trick…

Also, AFAICT there is no mention of i2c in the documentation until the troubleshooting section, it would be nice if the documentation mentioned this.