piface/PiFace-Real-Time-Clock #4: Add a license

nemequ, 2015-08-28:
I'm creating a recipe for yocto, and I need to put something for the license.

My preference would be the [MIT license](http://opensource.org/licenses/MIT), but it doesn't really matter.

tompreston, 2015-08-28:
Well the reason we didn't put one is because it's a fairly trivial script. Do you need me to add a license or do you just need permission to go ahead and use MIT? If so then go ahead! :)

nemequ, 2015-08-28:
It's fairly trivial, but still copyrightable :(.

I'm actually forking the code since I can't quite use it as-is, so I'll just slap an MIT license on it now that you've given permission, but it would still be a good idea to add a license so there is no issue for anyone else in the future.

closed 2016-04-29 by tompreston