piface/PiFace-Real-Time-Clock #5: Diagnosing PiFace-RTC hardware

gav-, 2015-09-13:
Hi, I can't locate another contact method to obtain hardware support for a PiFace-RTC, so I am posting here. I have been unsuccessful in getting mine to function, most recently attempted on latest raspbian on both a Pi2 B, and an older Pi rev1 B.

The problem is that is impossible to read or write to the RTC, but it does appear to be detected.

```
root@raspi:/home/pi# tail -n 6 /etc/rc.local 
modprobe i2c-dev
# Calibrate the clock (default: 0x47). See datasheet for MCP7940N
i2cset -y 0 0x6f 0x08 0x47
modprobe i2c:mcp7941x
echo mcp7941x 0x6f > /sys/class/i2c-dev/i2c-0/device/new_device
hwclock -s

root@raspi:/home/pi# hwclock -s
hwclock: ioctl(RTC_RD_TIME) to /dev/rtc0 to read the time failed: Invalid argument

root@raspi:/home/pi# hwclock -w
hwclock: ioctl(RTC_RD_TIME) to /dev/rtc0 to read the time failed: Input/output error
hwclock: ioctl(RTC_SET_TIME) to /dev/rtc0 to set the time failed.: Input/output error

root@raspi:/home/pi# i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- UU 
70: -- -- -- -- -- -- -- --                

root@raspi:/home/pi# echo 0x6f > /sys/class/i2c-dev/i2c-0/device/delete_device 
root@raspi:/home/pi# i2cdump -y 0 0x6f b
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: b3 02 01 39 01 01 01 80 47 01 00 00 00 01 01 01    ???9????G?...???
10: 01 03 00 00 01 01 01 01 34 00 01 21 34 00 01 21    ??..????4.?!4.?!
20: ef 31 48 26 38 bc 11 40 25 11 19 86 23 51 46 8d    ?1H&8??@%???#QF?
30: 8d 40 20 07 12 93 99 00 2f 02 09 24 9b ca 02 24    ?@ ????./??$???$
40: 90 a0 4a 21 09 0a 20 4e a0 02 82 39 24 11 a1 1a    ??J!?? N???9$???
50: c4 cc 06 32 00 00 1c 41 36 88 a1 60 42 30 86 40    ???2..?A6??`B0?@
60: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
70: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
80: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
90: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
a0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
b0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
c0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
d0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
e0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX
f0: XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX XX    XXXXXXXXXXXXXXXX    
```

Any help in diagnosing this issue would be appreciated.

tompreston, 2016-05-04:
Hi, sorry for taking so long getting back to you.

Address 0x6f is always UU, PiFace RTC should appear at 0x57. Do you have a battery in you RTC?

tompreston, 2016-05-04:
Hi,
Sorry it's taken so long to get in touch. Can you remove those lines from your `/etc/rc.local`, reboot and run the following commands:

    $ sudo su
    # modprobe i2c-dev
    # i2cset -y 0 0x6f 0x08 0x47
    # modprobe i2c:mcp7941x
    # echo mcp7941x 0x6f > /sys/class/i2c-dev/i2c-0/device/new_device
    # hwclock -r

Use `i2c-1` if you're on a Raspberry Pi 2 or above. Do you get a time read out?

gav-, 2016-05-05:
Thanks for getting back to me, I had all but given up on getting the PiFace RTC to work.

I have checked the RTC battery and it is reading 3.12v between battery positive and pi gnd.

I am testing on a Pi B rev1 (256MB RAM) with a fresh install of latest Raspbian Jessie with I2C enabled (dtparam=i2c_arm=on)

Following your instructions (I had to modprobe i2c-bcm2708 too):
```
pi@raspi:~ $ sudo su
root@raspi:/home/pi# modprobe i2c-dev
root@raspi:/home/pi# i2cset -y 0 0x6f 0x08 0x47
Error: Could not open file `/dev/i2c-0' or `/dev/i2c/0': No such file or directory
root@raspi:/home/pi# modprobe i2c-bcm2708
root@raspi:/home/pi# i2cset -y 0 0x6f 0x08 0x47
root@raspi:/home/pi# modprobe i2c:mcp7941x
root@raspi:/home/pi# echo mcp7941x 0x6f > /sys/class/i2c-dev/i2c-0/device/new_device
root@raspi:/home/pi# hwclock -r
hwclock: ioctl(RTC_RD_TIME) to /dev/rtc to read the time failed: Invalid argument
root@raspi:/home/pi# ls -la /dev/rtc*
lrwxrwxrwx 1 root root      4 May  5 16:18 /dev/rtc -> rtc0
crw------- 1 root root 254, 0 May  5 16:18 /dev/rtc0
```

```
# i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 6f 
70: -- -- -- -- -- -- -- --               
```

tompreston, 2016-05-05:
It appears you don't have access to `/dev/i2c-0`. Maybe you have `/dev/i2c-1`?

Try running:

    ls /dev/i2c-*

To see which I2C devices you have available.

gav-, 2016-05-06:
I think you may have missed my comment about needing to `modprobe i2c-bcm2708`, but it was the reason for the error about access to `/dev/i2c-0`. Please re-check my output, once I insert the module my call to `i2cset -y 0 0x6f 0x08 0x47` completes without error. 

(I have now removed `blacklist i2c-bcm2708` from the `/etc/modprobe.d/raspi-blacklist.conf` and I don't need to manually modprobe it any more, but the same final issue with failing to read /dev/rtc persists).

Output as requested:
```
root@raspi:/home/pi# ls /dev/i2c-*
/dev/i2c-0
root@raspi:/home/pi# grep 'Revision' /proc/cpuinfo
Revision	: 0003
```

tompreston, 2016-05-09:
It's so strange that `hwclock` doesn't seem to be working even though you get `/dev/rtc` after loading all of the modules.

Hm, what do you get out of:

    # hwclock --debug

I get:

    # hwclock --debug
    hwclock from util-linux 2.25.2
    Using the /dev interface to the clock.
    Assuming hardware clock is kept in UTC time.
    Waiting for clock tick...
    /dev/rtc does not have interrupt functions. Waiting in loop for time from /dev/rtc to change
    ...got clock tick
    Time read from Hardware Clock: 2016/05/09 13:10:42
    Hw clock time : 2016/05/09 13:10:42 = 1462799442 seconds since 1969
    Mon 09 May 2016 13:10:42 UTC  -0.791762 seconds

gav-, 2016-05-11:
> It's so strange that hwclock doesn't seem to be working even though you get /dev/rtc after loading all of the modules.

Right. That's been the symptom from day one (tested on different Raspberry Pi hardware), so I've always suspected that the PiFace-RTC was DOA.

The command `echo mcp7941x 0x6f > /sys/class/i2c-dev/i2c-0/device/new_device`, shows the following in dmesg:
```
[  106.413544] rtc-ds1307 0-006f: SET TIME!
[  106.415192] rtc-ds1307 0-006f: SET TIME!
[  106.418350] rtc-ds1307 0-006f: rtc core: registered mcp7941x as rtc0
[  106.418415] rtc-ds1307 0-006f: 64 bytes nvram
[  106.418484] i2c i2c-0: new_device: Instantiated device mcp7941x at 0x6f
```
So /dev/rtc is created as expected, but hwclock always fails. Debug output as requested:
```
root@raspi:/home/pi# hwclock --debug
hwclock from util-linux 2.25.2
Using the /dev interface to the clock.
Assuming hardware clock is kept in UTC time.
Waiting for clock tick...
/dev/rtc does not have interrupt functions. Waiting in loop for time from /dev/rtc to change
hwclock: ioctl(RTC_RD_TIME) to /dev/rtc to read the time failed: Invalid argument
...synchronization failed
```

ambrosa, 2016-08-09:
Hi Guys.

Same problem here.
I've just bought a PiFace Clock and I've installed into an old RPI model A (256MB RAM) with bus i2c-0
A Raspbian Jessie Lite fresh installation.

PiFace Clock module has worked fine for about 30 minutes.

Then after a reboot
_root@Pi:~# hwclock -r
hwclock: ioctl(RTC_RD_TIME) to /dev/rtc to read the time failed: Invalid argument_

Very strange. Ant it not worked any more. I've removed the battery CR1220, I've reinserted the PiFace Clock module. Nothing. It doesn't work anymore.
Dead ?

ambrosa, 2016-08-09:
dmesg looks to be correct

_[    7.911265] bcm2708_i2c 20205000.i2c: BSC0 Controller at 0x20205000 (irq 77) (baudrate 100000)
[   10.986294] rtc-ds1307 0-006f: SET TIME!
[   10.987901] rtc-ds1307 0-006f: SET TIME!
[   10.993322] rtc-ds1307 0-006f: rtc core: registered mcp7941x as rtc0
[   10.993388] rtc-ds1307 0-006f: 64 bytes nvram
[   10.993501] i2c i2c-0: new_device: Instantiated device mcp7941x at 0x6f_

stovalldb, 2016-08-22:
Did any of you resolve the PiFace RTC issue? I purchased one some time back and just got back to a project. I need to be able to Date/Time stamp bluetooth pairing. Pi will be offline.
Footnote: just learning Python, last programming experience was with Fortran and Punch Cards, if you say, "what's Fortran or Punch Cards" that was early '70s

gav-, 2016-08-22:
Hi stovalldb,

By the time I confirmed that my PiFace-RTC was dead, I gave up on it. It was unlikely the supplier (Element 14) would give warranty on it by then. I soured on the using them after that.

You never mentioned if you have the same problem with yours. If so, contact your supplier.

I was still taught Fortran 77 at University in the early '90s (just without the punch cards).

stovalldb, 2016-08-22:
Gavin,

Thanks for your reply. As mentioned I just re-started a project so pulled out my B and loaded Jessie and installed the PiFace RTC. The instructions on their website using wget didn’t work. So spent a lot of hours just trying to find and get some software installed. Getting ready to order from Amazon:  <https://www.amazon.com/gp/product/B01AT239Z6/ref=ox_sc_act_title_2?ie=UTF8&psc=1&smid=A2IAB2RW3LLT8D> 
UCTRONICS DS1307 RTC I2C Real Time Clock Module High Precision for Raspberry Pi 2 model B Arduino by UCTRONICS. Hoping to have better luck with it.

I am trying to replace a gate opener control board that came over on the Mayflower. Will install a Pi at the gate and have my customers pair their smart phones with the Pi using a Key and trip an output to open the gate. Really want to Date/Time stamp the pairing for historical records. The Pi will only be online to access the historical records thru a HotSpot.

Have you tired any other RTC with any success?

David

stovalldb, 2016-08-22:
Gavin,

Is the Raspberry Pi a hobby or do you make a living programming?

David

gav-, 2016-08-23:
If your issue is different to mine (and it sounds like it might be), then I would strongly suggest you open a new issue about it, and include details about the commands you have used, including the errors you are seeing.

I haven't used any other RTC with the pi, so I can't suggest one.

Yes, I write software for a living, but I only use the Raspberry Pi in a hobby capacity (for the moment anyway).

stovalldb, 2016-08-23:
The first issue was the instructions from PiFace for the download don’t work. The link is wrong.  All your posts helped some but being a beginner was just a little over my learning curve right now. I will continue to work on my Bluetooth side of the project and come back to the RTC later.

fc-gituser, 2016-11-14:
Dear Sir, I have been trying to install PiFace Real Time Clock on a Raspberry Pi Zero (Jessie distribution) for a number of months but have been unsuccessful. I have many of these devices and would llike to install them on RPi Model 3 as well when issues are resolved. I had installed a number of these devices over a year ago through /etc/rc.local but now that a new revision is out using /etc/init.d/ pifacertc these devices are just not identified the "6f" code is shown and when the driver is loaded it shows the "UU". However, a vast number of attempts of invoking sudo ./install-piface-real-time-clock.sh results in a message which states that an additional init script needs to be created but never is explained or described in detail. All variations of code developed and tried in an element14 community discussion- https://www.element14.com/community/message/207318/l/re-installing-code-for-piface-real-time-clock#207318 - results in " Failed to install the PiFace RTC. It may be that the install script in raw.github.com/piface/PiFace-Real-Time-Clock does not test for RPi ZERO's. 
In any event, below is the entire discussion of the issue I have been confronted with and unfortunately I do not had enough knowledge to resolve the issue.

Installing Code for PiFace Real Time Clock 

This question is Not Answered.(Mark as assumed answered)

Frank Caracciolo 
Level 1: Coulomb
 
Frank Caracciolo Oct 11, 2016 4:52 AM 

Dear Element 14 Community.

All my Raspberry Pi devices (RPi 2, RPi Zero, and RPi 3 (planned) are running with the current Raspbian Jessie version. I have successfully installed PiFace Real Time Clock using the PiFace website instructions and GitHub.com software download on a number of RPi 2's and one of my three RPi Zero's.

 

I have recently continued the installations on another two RPi's and cannot continue because after these two lines:

1. chmod +x install-piface-real-time-clock.sh

2. sudo ./install-piface-real-time-clock.sh

 

The following new comments are returned: 1

1.. create new pifacertc init script to load time from PiFace RTC

2. Adding /etc/init.d/pifacertc .

3. install the pifacertc init script

4. enable I2c by using raspi-config

 

As I am a Linux beginner, can someone lead me through (expand) the coding of these lines so I may install PiFace RTC currently and in the future. The changelog on GitHub refers to a revision change but I am not knowledgable enough to understand why the install went through previously by not currently.

 

Thank you very much,

Frank.
 

I have the same question 
Show 0 Likes
(0)  

818 Views  
 
Tags (edit)

:      
Reply 
 '

 

Average User Rating: No ratings (0 ratings)

Average User Rating

No ratings

(0 ratings)

Your Rating: 
[... 2419 lines cut ...]
I never got a positive response before.

 

I guess that's all I did.

I hope I gave you some further information. It appears that the RTC is being identified but not completely.

 

The command

echo mcp7941x 0x6f > /sys/class/i2c-dev/i2c-1/device/new_device

permission denied disturbed me

so I added a sudo in front of echo but it returned with something like "bad command".

I need to get the correct text allow permission.

 

Your consideration is of utmost value,

thanks, Frank.