piface/PiFace-Real-Time-Clock #12: Wrong i2c bus number on Raspberry Pi model B+

dalathu, 2017-12-16:
Hello,

I have used the new installation script (commit 056305b7d383fbde63342de9427f6538d7f0e415) in my Raspberry Pi model B+. It does not work due to that the `pifacertc` script tries to configure the wrong i2c bus number, i.e. 0 instead of 1.
On my Raspberry pi, the i2c bus device is `/dev/i2c-1`. 

Output of `/proc/cpuinfo`:
```
$ grep "Revision" /proc/cpuinfo  
Revision        : 0010
```
I have solved the issue by modifying the 2 following lines in the file `/etc/init.d/pifacertc`:
```
i2cset -y 1 0x6f 0x08 0x47
```
and
```
echo mcp7941x 0x6f > /sys/class/i2c-dev/i2c-1/device/new_device
```

csoutreach, 2017-12-16:
Thanks -- we'll be posting an updated version that should fix this soon.

sadilekivan, 2020-10-31:
Hello,

The same issue occurs on a Raspberry Pi Zero W V1.1 @2017, changing to i2c-1 worked, just wanted to add it here as a note.

Thanks for the script, I was firstly discouraged and tried many other solutions, which did not work in the end. But after all that tinkering I learned a lot and realized what the issue was.